Dlloydev/jtag2updi #1: Trouble uploading using Arduino Nano HV UPDI Programmer

Nbelles, 2020-08-05:
Hi,

I recently purchased two of your "Arduino Nano HV UPDI Programmer" from your Tindie shop and just got them in the mail and have been trying to get them up and working. I started by just trying to write a simple blink sketch to the ATtiny412 and sometimes it works and other times it doesn't even though I haven't changed any of the wiring or code. Sometimes I get one set of errors and other times I get other errors and sometimes I don't get any, the one that I get most often is shown below. If I get the other error message again, I'll try and add it to this reply. My current setup includes and Arduino Nano plugged into the Arduino Nano HV UPDI Programmer and the Write Protect jumper plugged in and the PCHV UPDI jumper plugged in. Then the three UPDI pins are plugged directly into the ATtiny412. Thanks for any help!

Here are the most common error messages: 

```
Sketch uses 716 bytes (17%) of program storage space. Maximum is 4096 bytes.
Global variables use 10 bytes (3%) of dynamic memory, leaving 246 bytes for local variables. Maximum is 256 bytes.
/Users/Nathaniel/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/Nathaniel/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5/avrdude.conf -v -pattiny412 -cjtag2updi -P/dev/cu.usbserial-1410 -Uflash:w:/Users/Nathaniel/Documents/Arduino/build/Blink.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/Nathaniel/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5/avrdude.conf"
         User configuration file is "/Users/Nathaniel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATtiny412
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no       4096   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          00:00:00:00:00:00
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_NO_TARGET_POWER
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_NO_TARGET_POWER
avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_NO_TARGET_POWER
avr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude: error reading signature data for part "ATtiny412", rc=-2
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_NO_TARGET_POWER

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Dlloydev, 2020-08-05:
Thank you for your order ... lets get these up and running.

**Re: TARGET**
What board are you using or is this a breakout module?
Does your ATtiny412 have a decoupling capacitor (usually 0.1µF) installed close to the VCC and GND pins?
Make sure the only connections are:
1. UPDI, direct connection to SOIC pin 6 (PA0), **no series resistor** needed as this is built in to the programmer.
2. GND
3. VCC (T5V) from programmer (make sure no other source of power is connected)

**Re: Arduino Nano and Programmer PCB**
Are you using a Nano clone?
Follow the Assembly and Setup procedure [here](https://github.com/Dlloydev/jtag2updi/wiki/Arduino-Nano-HV-UPDI-Programmer).
With the jumpers configured as you've described, you'll notice the programmer's blue HV LED blink once just after power is applied (USB cable connected).

**Software:**
What version of Arduino are you using ... Windows Installer, Windows App or other?
I've just run some tests using an ATtiny412 dev board from Azzy's Electronics on tindie. My PC is using Arduino IDE 1.8.13, megaTinyCore 2.05 and Windows 10 ... couldn't get any errors.
For testing, here's the [board setup](https://i.imgur.com/bXZfVhz.png) I used. (also tested with UPDI as Reset and GPIO).
Blink sketch: 
```
void setup() {
  PORTA.DIRSET |= PIN7_bm;
}

void loop() {
  PORTA.OUTSET = PIN7_bm;
  delay(50);
  PORTA.OUTCLR = PIN7_bm;
  delay(300);
}
```
After setting up the board or making any changes its options, run "Burn Bootloader"
Then upload your sketch using "Upload Using Programmer".

Nbelles, 2020-08-05:
**Re: TARGET**
I am using the ATtiny412 soldered onto an SOIC-8 to DIP-8 converter that is plugged into an empty breadboard (no other connections than the three you mentioned). I do not have a capacitor across the supply voltage but I will try that right now. I put the capacitor in and that seemed to solve the problem. But then once I unplug the capacitor (to make sure I'm not crazy) it still is working! I'm not exactly sure what's going on but it seems like it is working now to program normally. Now, the reason I purchased this high voltage programmer is because I want to be able to use the UPDI pin as an output and (even with the capacitor in there) I am still not able to upload a program that blinks on the UPDI pin. 

**Re: Arduino Nano and Programmer PCB**
I'm not sure if it is a clone or not, someone gave them to me for free. Is there an easy way to check and see if it is a clone or not? I followed all the steps to get the jumpers in the right place for PCHV and WP. I am getting a blue LED flash very quickly when I plug in the Nano to my computer so I think that means I am in the right programming mode. 

**Re: Software:**
Arduino 1.8.10 for Mac with megaTinyCore 2.0.5 and macOS 10.15.6
I just tested your blink sketch with my setup and it does work properly. To try using the UPDI pin as an output I tried changing PIN7_bm to PIN0_bm to hopefully get that working but even though it compiles and uploads, it doesn't functionally work. It just stays HIGH all the time. I even tried unplugging the UPDI pin and power to the ATtiny412 to see if that would change anything but it didn't, it still just stays HIGH. 

Thanks for the quick reply and at least getting the programmer working. Now we just gotta get the UPDI pin setup as an output!

Dlloydev, 2020-08-05:
Thank you for your reply ... glad its working!

Now for UPDI as GPIO.
- I would first of all change the BOD voltage to 4.2V for now.
- For your board, you'll need to select this option [here](https://i.imgur.com/nto2tKX.png) (optiboot) to allow changing the UPDI configuration.
- Now you can select UPDI/Reset pin: IO DANGER "Read docs first! powercycle to enter..." Note that with your existing mode setting of the programmer, the powercycle will be automatically performed.
- Run "Burn Bootloader"
- In the sketch, change PIN7_bm to PIN0_bm
- Now make sure the LED circuit is disconnected from PA0
- Upload with "Upload Using Programmer", reconnect the LED, and success?
- With the UPDI pin as an output, remember to disconnect the LED prior to any programming session.

Nbelles, 2020-08-05:
It works!

Thank you so much for your help. A couple of other quick questions:

- Does using Optiboot take up more storage on the chip than not using Optiboot?
- Are there any other implications of using Optiboot as opposed to not? I assumed that it was better to not use Optiboot unless I absolutely needed to. (In all of Dr. Azzy's megaTinyCore documentation he says that Optiboot is not recommended even though he supports it)? 
- Theoretically, I should always be able to reprogram/recover any ATtiny using the PCHV option on the programmer no matter whether I'm using Optiboot or not. Is that correct?
- Are there any downsides to always using the PCHV option? Will I eventually damage the chip if I always program with PCHV?

Dlloydev, 2020-08-05:
Yes, optiboot does take up some storage on the chip, probably why its not recommended on the 8-pin chips because there's only 4KB max available, however if your code fits, no problem! Optiboot will also allow for programming through serial (RX/TX) so then you could use a serial adapter. I'm quite sure you could use avrdude directly using a terminal window and change fuse5 to get any desired setting for the UPDI. In this case, you wouldn't need optiboot and get some extra memory. I think the UPDI configuration isn't available (yet) for non-optboot configurations so that users don't inadvertently get locked out of the MCU and find out they don't have a HV programmer.

Yes, the PCHV (Power Cycle High Voltage) option can be used no matter whether your using Optiboot or not. It also doesn't matter how the UPDI pin is configured (UPDI, Reset or GPIO). Only need to disconnect you user circuit - LED. 

There's no downside to always using the PCHV option, it will have no effect on the chips life expectancy or reliability as this pin is rated for 13V max. Note that the HV signal is a quick 12V pulse only 700µs in duration. In PCHV mode, A HV pulse is applied when power is connected.  Then,  anytime you program (burn bootloader or upload) a power cycle is performed and a second HV pulse is sent. Just after programming completes, a second power cycle occurs to re-load the fuses. 

Note that for the HV mode of the programmer, it will only apply a HV pulse at power-up. No power cycles occur. This would be used when the target circuit is used to power the programmer for in circuit programming.

Nbelles, 2020-08-05:
Sounds good. Do you know if Dr. Azzy is working on adding UPDI configuration for non-Optiboot based bootloaders in Arduino? I don't really want Optiboot on these 8-pin chips (because you're right, they don't have a lot of storage) but I would like to be able to use that last pin as GPIO and just program using PCHV. I don't really know a lot about fuse setting (except that you have to "burn bootloader" through the Arduino IDE to have them take effect), would you be able to walk me through that? 

I'm willing to add an issue/feature request to his repo if you don't know.

Dlloydev, 2020-08-05:
He has an issue(enhancement) opened [here](https://github.com/SpenceKonde/megaTinyCore/issues/161)
I'll try using avrdude later this evening and report back.

Nbelles, 2020-08-05:
I just did some looking through Dr. Azzy's boards.txt for megaTinyCore hoping that he might set all the fuses manually when options are selected for GPIO or Reset and I see this:

```
atxy2o.menu.resetpin.gpio=IO (DANGER - Read docs first! powercycle to enter bootloader)
atxy2o.menu.resetpin.gpio.bootloader.file=optiboot_x/optiboot_txy2{bootloader.uartswap}.hex
atxy2o.menu.resetpin.gpio.bootloader.resetpinbits=00
```

Which makes me think that there is a lot more in that .hex file that affects the pins besides just the "resetpinbits=00" but again, I'm way over my head and don't want to trash some of my ATtiny's messing around with it.

Dlloydev, 2020-08-05:
OK, I'll put the avrdude procedure on hold.
Note: If you use the UPDI pin as Reset, you can leave the reset push-button connected while programming.
Good luck with your testing!

Nbelles, 2020-08-05:
I would love to see if it is possible just with avrdude whenever you have some free time. In the meantime I'll be on the lookout for a resolution on the enhancement you mentioned in megaTinyCore. 

All your help has been greatly appreciated!

closed 2020-08-05 by Nbelles