Dlloydev/jtag2updi #3: Possible to run on ATtiny?

Nbelles, 2020-08-15:
Hey,

I've been looking through the code to see if it is possible to run this code on an ATtiny (1614 for instance) and was curious if you had any input. I would love to run this code on an ATtiny with a FTDI chip (or any other USB to Serial converter). I see that there are a lot of instances of it mentioning being compiled on Series-0 and Series-1 tinyAVR devices. I understand that your modifications use Port C (on the ATmega devices) and I don't really need that sort of feedback for my implementation. Just curious what you think.

Thoughts?

Dlloydev, 2020-08-15:
Yes, I've had a [version of this firmware running a on an ATtiny1604](https://forum.arduino.cc/index.php?topic=669577.msg4561698#msg4561698) (with optiboot) and used an FTDI LC231X Module for the serial connection to the PC. For target power cycle, I used an external power switch. The Nano HV Programmer uses Port C on the ATmega328P for target power and power-cycle. I didn't release anything as I've decided to opt for the AVR DA for a more advanced programmer with the capability of setting the target voltage from 3V-5V in a few mV per step.  Neat thing is that the AVR DA could also operate from 3V-5V without changing the MCU frequency, which saves level translator switching components.

In order to use the ATtiny1614 as a programmer,`#if defined (__AVR_ATtiny1614__)` wrappers around the required programmer functions and code blocks need to be added to the firmware.

Nbelles, 2020-08-16:
Is there any reason you used Optiboot? Is it needed for something that I'm not thinking about?

Dlloydev, 2020-08-16:
Optiboot allowed programming with the FTDI serial adapter (like an Arduino) and using the UPDI pin configured as Reset, which is connected to an auto-reset circuit and DTR. When auto-reset is disabled, the ATtiny with jtag2updi firmware operates as a programmer. When auto-reset is enabled, the programmer's firmware is updated during an upload. 

You could avoid optiboot if you dedicate an extra pin configured with a reset interrupt function.

Nbelles, 2020-08-16:
I was thinking I could just keep the UPDI port for uploading. 

The problem I am having is that even though I think I've made all the proper changes to the code, it doesn't seem to respond to anything. I got a bunch of ` avrdude: jtagmkII_getsync(): sign-on command: status -1 ` errors when trying to program another ATtiny with it. I put an oscilloscope probe on the TX and RX pins (between the FTDI chip and the ATtiny programmer) but it seems that it doesn't respond. I've attached a picture of the oscilloscope output on the FTDI chip. Not sure if that is helpful. 

[image]

Dlloydev, 2020-08-16:
Yeah, data flow back to the serial port won't happen until after sign-on. 

I have all the pieces tested that would match up with the features of the Nano HV UPDI Programmer ... so I'll do some copy and paste from my various revisions and come up with something to publish. This should be easier to work with for your implementation. However, may need a week at the most.

Nbelles, 2020-08-16:
Thanks so much. I can't wait!

Dlloydev, 2020-08-18:
Its ready ... see Wiki for documentation.

[image]

Nbelles, 2020-08-18:
Sweet! I'll try it out tonight. How did you create that pinout chart? I've seen them all over but want to know how they are made.

Dlloydev, 2020-08-18:
Just an original pinout edited with Paint 3D. I stumbled upon an open source application and youtube video a while back, but can't think of what its called.