Dlloydev/jtag2updi #7: ST662ACD charge pump is now obsolete

Dlloydev, 2020-08-29:
Received Part Life Notification from Digi-Key for the ST662ACD charge pump. Unfortunately, it has been marked with Obsolete status and is now non-stock. Since now there's no low cost directly compatible solution, this puts higher priority on the 4-pin built-in charge pump solution.

Since there are enough pins available on the 14-pin ATtiny, I'll be adding back the built-in charge pump as follows:

On the 20 and 24-pin parts, CPS (charge pump shutdown) will moved from PB4 to PB1.
On the 14-pin part, CPS will be on pin PB1, CP1 added to PA4 and CP2 added to PA5.

In summary, for all ATtiny parts, we will have:

Pin | Charge Pump pin name
-- | --
PA3 | CPP / HV Enable
PA4 | CP1 charge pump
PA5 | CP2 charge pump
PB1 | CPS charge pump

Dlloydev, 2020-08-30:
Update completed.

closed 2020-08-30 by Dlloydev

ssarna1, 2023-03-26:
FYI: MAX662ACSA is a direct and non-obsolete replacement.