Dlloydev/jtag2updi #9: Question for AVR-DA

sstaub, 2020-09-13:
Can the programmer also used with DxCore (AVR-DA and AVR-DB) and MegaCoreX (AtMega4809...)?

Dlloydev, 2020-09-13:
Good question. I've used the Nano Every with MegaCoreX  and got to the point of everything working except the serial. I needed to do more investigation with the built-in SAMD programmer chip and UPDI line. 

Should be no issues adding full HV programming support to the 4809 on its own ... I'll do this first prior to adding support for the AVR-DA/DB with DxCore as the code would be quite similar. So to answer your question ... not yet, but coming soon!

sstaub, 2020-09-13:
Thank's for the fast response, hope AVR-DA support will follow, I  ordered some of 28-pin DIP one direct from microchip, also 2 of your programmers from Tindie. Meanwhile I can use the code from SpenceKonde which should also work with the Nano, but with no HV.

Dlloydev, 2020-09-14:
Added HV support for ATmega4809 Family
This includes the ATmega4809, ATmega4808, ATmega3209, ATmega3208, ATmega1609, ATmega1608, ATmega809 and ATmega808 devices. Pinout diagrams and new information added to the GitHub Wiki.

Also added HV support for the AVR DA, but getting compile errors due to missing pin definitions ... left an issue at the DxCore repository.

Dlloydev, 2020-10-02:
New update with ATtiny and ATmega HV Programmer fixes, updated documentation with working 40-pin ATmega4809 on breadboard.

Dlloydev, 2020-10-11:
I'll be watching for this upstream.

Dlloydev, 2020-12-06:
Taking another look at this...
Should be able to add HV programming support for AVR-DA parts with 128K flash memory
(AVR128DA32, AVR128DA48 and AVR128DA64).

Dlloydev, 2020-12-06:
Done!
Repo updated.
For pinout descriptions, just refer to the 28/48/64 pin images for the [DIY ATmega4809 HV UPDI Programmer](https://github.com/Dlloydev/jtag2updi/wiki/DIY-ATmega4809-HV-UPDI-Programmer).

closed 2020-12-06 by Dlloydev