Dlloydev/jtag2updi #11: No target power

nieldk, 2020-12-07:
Since Latest commit, i get an error, No target power.

If i move 5V from the updi header, to 5V on the arduino ISP header, it works, and flashes the attiny412 (need 2 tries, as it should atm)

Using your programmet for the nano, from Tindie.

nieldk, 2020-12-07:
Hmm seems to have fixed itself.
Closing this

closed 2020-12-07 by nieldk

Dlloydev, 2020-12-07:
Thanks for reporting this.

I ran some tests on this repo and the one previous to enabling pullups for unused pins and the AVR132DA ... the results came out the same. I also tried various chip settings, but it had no effect on the results, so I used defaults.
 
I updated the [Arduino Nano HV UPDI Programmer page](https://github.com/Dlloydev/jtag2updi/wiki/Arduino-Nano-HV-UPDI-Programmer) to show **Operational Test Results** (scroll down near the bottom).

Let me know if you're finding any differences or new issues.

nieldk, 2020-12-07:
> Thanks for reporting this.
>
> I ran some tests on this repo and the one previous to enabling pullups for unused pins and the AVR132DA ... the results came out the same. I also tried various chip [...]

Thank you.

I have been able to reproduce the error situation. After flashing this code, the attiny412 gets in a state where I need to supply the 5V from 6 PIN header instead of UPDI header (T5).```

```
#include <avr/io.h>
#include <util/delay.h>

int main(void) {
  PORTA.DIR = PIN3_bm;

  while(1) {
    PORTA.OUTTGL = PIN3_bm;
    _delay_ms(500);
  }
}
```

[image]

Flashing this code, brings the expected functionality (T5) back again
```
#include <avr/io.h>
#include <util/delay.h>

#define LED_PIN PIN3_bm
#define LED_PORT PORTA

void setup() {
  LED_PORT.DIRSET = LED_PIN;
}

void loop() {
  if ((FUSE.SYSCFG0 & FUSE_RSTPINCFG_gm) == RSTPINCFG_UPDI_gc) {                     // UPDI mode
    ((millis() >> 7) & 15) ? LED_PORT.OUTCLR = LED_PIN : LED_PORT.OUTSET = LED_PIN;  // LED blinks at 0.5Hz
  }
  else if ((FUSE.SYSCFG0 & FUSE_RSTPINCFG_gm) == RSTPINCFG_RST_gc) {                 // Reset mode
    ((millis() >> 5) & 15) ? LED_PORT.OUTCLR = LED_PIN : LED_PORT.OUTSET = LED_PIN;  // LED blinks at 2Hz
  }
  else if ((FUSE.SYSCFG0 & FUSE_RSTPINCFG_gm) == RSTPINCFG_GPIO_gc) {                // GPIO mode
    ((millis() >> 3) & 15) ? LED_PORT.OUTCLR = LED_PIN : LED_PORT.OUTSET = LED_PIN;  // LED blinks at 8Hz
  }
  else {                                                                             // Invalid setting
    LED_PORT.OUTSET = LED_PIN;                                                       // LED remains ON
  }
}
```
[image]

Dlloydev, 2020-12-07:
I tested your fist code and couldn't duplicate the issue until I forgot to switch the programmer from HV Mode to UPDI Mode. Although leaving the programmer's jumper in HV mode or PCHV mode doesn't hurt the target's PA0 when its in UPDI mode, this can start the "toggle" pass/fail issue. Also, note that the T5V pin on the programmer comes from PORTC and is only good for about 60mA and its voltage is about 4.7-4.8V.

For simple UPDI programming, I recommend having the jumper in UPDI mode and using the 5V pin on the ISP header.
This jumper is difficult to access ... I think the long flexible tab versions would be easier to use.

I'll update the product page to show recommended connections and jumper settings.

From your images, it looks like you're using an older version of megaTinyCore.
I'm using 2.1.5 installed via Boards Manager which does show a few compiler warnings (these can be ignored) ... the GitHub version has this fixed.

nieldk, 2020-12-08:
Update megaTinyCore, still I can reproduce. I guess its the chip being used. I really dont know, the top has the text removed.
Examining the 12V pulse, seems OK, actually goes to ~16V very short.
Then again, I noticed you mentioned the use of the 6PIN header, so its ok really.
[image]

Thanks.

Dlloydev, 2020-12-09:
The voltage should be clipped to 12V by the zener D4 ... looks like you may have a bad solder connection or faulty diode.
I'll be happy to send a replacement asap (even if you manage to repair it).

[image]

nieldk, 2020-12-09:
Excellent, this looks more correct
[image]