Dlloydev/jtag2updi #12: Atmega 2560

grogcw, 2021-02-03:
Hi !

I'd like to use your schematics and code to make a HV/PCHV with an Arduino Mega 2560.

I think that the only changes to make are in sys.h for pin definitions.

Do I have to tinker only there or is there somewhere else to define things for this to work ?

Thanks for taking care,

Cheers !

Dlloydev, 2021-02-03:
OK, I've uploaded a commit that should give you a head start (untested).
Let me know if there's any issues or if you'd like to make any changes.

Here's the pinout diagram:

[image]

grogcw, 2021-02-03:
Hi !

Thanks for your quick reply !

I don't have the electronics to make any charge pump at the moment, but I think I could rig something with a 12v power supply with a tension divider.

Anyway, I'll keep you in touch !

Cheers !

closed 2021-04-30 by Dlloydev