Dlloydev/jtag2updi #13: Curious behaviour

iansexton, 2021-02-08:
I have one of your boards. At first it seemed fine. I didn't actually do anything I couldn't do already but it gave me the confidence to set the UPDI pin as reset. http://myiot.co.uk/attiny/
However, when I go go back & try to re-assign the pin it doesn't work. The chip (1614) is completely screwed. It won't work with optiboot or UPDI.
I half a decent Tektronix DSO so I had a look. I couldn't capture **any** HV pulse which confuses me.
I wonder if you have any debug firmware? It would be nice to know (eg) if the charge pump is working. As everything is on the 'wrong' side of the board it's quite tricky to debug :(

Dlloydev, 2021-03-04:
Oh gosh, my sincere apology for taking so long to reply! I've been trying to get a window of opportunity to do a small run of these but it's just not happening ... ugh ... I guess I'll need to do update on Tindie where there's a wait list. 

I do have a working unit I'll send to you ASAP!

The HV pulse should occur at about 10 milliseconds after USB power-up (when in HV or PCHV mode). For your bricked 1614, try connecting to solid 5V from the 2x3-pin header ... see Operational Test Results and workaround [here](https://github.com/Dlloydev/jtag2updi/wiki/Arduino-Nano-HV-UPDI-Programmer).

Dlloydev, 2021-04-29:
I hope it all worked out ... closing for now but please open an issue if there's anything outstanding you wish to resolve.

closed 2021-04-29 by Dlloydev