Dlloydev/jtag2updi #15: In this diagram, it is not clear where to connect tinyAVR?

P55-rgb, 2021-04-28:
In this diagram, it is not clear where to connect tinyAVR?
Can you add it (tinyAVR) to the diagram?
https://raw.githubusercontent.com/wiki/Dlloydev/jtag2updi/images/NanoProgSch.png
https://oshpark.com/shared_projects/cxgBKsCQ

Dlloydev, 2021-04-29:
From the diagram, its just the 3 connections here, where T5V is the target 5V power (this comes from A0-A5).
[image]
For steady 5V power, this can be taken from the 6-pin connector on the Nano (see Arduino's documentation to identify GND and 5V on this connector.

P55-rgb, 2021-04-29:
[image]
Two questions.
1. HVUPDI = 12v?
2. Pin 6=UPDI on tiny412, where to connect?

Dlloydev, 2021-04-29:
Yes, the HVUPDI signal will have a fast 12V pulse (about 400µs) just after power-up, but it will normally be 5V. Pin PA0 (UPDI) is 12V tolerant.

[image]

P55-rgb, 2021-04-29:
According to the documentation on tiny412 page 385, a 12V pulse should appear on the UPDI pin.
I don't understand how your HVUPDI circuit generates a 12 volt pulse?
Explain the scheme...

Dlloydev, 2021-04-29:
Its a Dickson Charge Pump ... explained in the [documentation](https://github.com/Dlloydev/jtag2updi/wiki/Arduino-Nano-HV-UPDI-Programmer) (scroll down near the bottom).

P55-rgb, 2021-04-29:
[image]
LEDs: Red, yellow, blue.
There is only one LED1 in the diagram.
Is LED1, this is: red, yellow, or blue? )))
Where are the LEDs installed (red, yellow, blue) and what is their designation?

Dlloydev, 2021-04-29:
The red and yellow led are on the Nano. LED1 is the blue LED which indicates the 12V pulse, this is on the programmer PCB. There's [images](https://raw.githubusercontent.com/wiki/Dlloydev/jtag2updi/images/NanoProgTop.png) available at the bottom of the documentation link, gerber files are available at Osh Park.

P55-rgb, 2021-04-29:
https://oshpark.com/shared_projects/cxgBKsCQ
How much will it cost to deliver your programmer to Russia, to Tomsk?

Dlloydev, 2021-04-29:
Sorry, I'm not building and selling these as I don't have the necessary equipment (never did receive the reflow oven purchased just prior to the pandemic). I'll need to update the info at the Tindie store.

closed 2021-04-29 by Dlloydev