Dlloydev/jtag2updi #17: Probable bug

dduehren, 2021-06-05:
I have some bricked ATtiny 1604s that I have been trying to unbrick using your HV/PHV programmer with JTAG2UPDI. 
I am not sure how some of these devices got bricked in the first place,  They were programmed in normal UPDI mode, and then became bricked over time.  My programming success improved once I made a nano based dedicated programmer.   That programmer bricked at device because I had selected 1614 instead of 1604, even though the programmer detected the mismatch in IDs.

Now I am observing the following behavior, using the latest version of the code (which  BTW seems to have fixed  ...sign-on command: status -1 problems)
1. On most devices, when power is attached, it still reports RSP_ILLEGAL_MCU_STATE as expected.
2. In an board where the ATtiny is embedded, and where it was initially programmed (with the wrong ID), the behavior is the opposite. In  Mode: PHV or HV.  Power connected to device, but report is RSP_NO_TARGET_POWER.   If I remove power, I get RSP_ILLEGAL_MCU_STATE.  

Any thoughts on what could cause this result.  Could it explain why the device was bricked in the first place by selecting the wrong chip?

Dlloydev, 2021-06-05:
Its been awhile since I've used any HV programmer and I see megaTiny core is now at v2.3.2. Anyways, no matter what HV programming circuit or board you're using or how the 1604 became bricked, here's some tips to bring it back:

**Preparation:**

- Use only 3 connections to the target IC or board UPDI, GND and direct 5V power (not T5V which is lower 4.8V and power cycles).
-  For the UPDI signal, connect directly to PA0 (UPDI) on the 1604 No extra series resistance is needed and is not beneficial (the HV programmer will already provide some resistance and a diode to protect its own UPDI pin).
- Make sure there are no other connections i.e serial which will back-feed power through the IO pins.
- Make sure the 1604 has bypass capacitor from VCC to GND.
- Make sure there is some load on the 5V power (220R-1K) so the programmer and target will properly discharge when disconnected.
- Use a DC adapter powered USB hub to ensure proper and more stable voltage to the programmer and target.
- Set the programmer to HV mode. When the USB power is connected,  a 12V pulse will be applied the target's UPDI pin.

**HV Programming**

- HV Mode: Connect USB power → 12V pulse,  → "burn bootloader", → if fails, "burn bootloader" again.
- PCHV Mode:  Connect USB power → 12V pulse,  → (12V pulse  → "burn bootloader") → if fails, (12V pulse  → "burn bootloader") again. 
- Note the extra 12V pulse occurs just before programming and just after the programmers power-cycle which isn't connected. The extra 12V pulse(s) are not harmful. 

Here's the latest test results I had using a Nano HV UPDI Programmer with Attiny412 target:

<h3 cid="n2" mdtype="heading" class="md-end-block md-heading md-focus" style="box-sizing: border-box; break-after: avoid-page; break-inside: avoid; orphans: 4; font-size: 1.5em; margin-top: 1rem; margin-bottom: 1rem; position: relative; font-weight: bold; line-height: 1.43; cursor: text; white-space: pre-wrap; color: rgb(51, 51, 51); font-family: &quot;Open Sans&quot;, &quot;Clear Sans&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-style: initial; text-decoration-color: initial;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">Operational Test Results</span></h3><ul class="ul-list" cid="n3" mdtype="list" data-mark="-" style="box-sizing: border-box; margin: 0.8em 0px; padding-left: 30px; position: relative; color: rgb(51, 51, 51); font-family: &quot;Open Sans&quot;, &quot;Clear Sans&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-style: initial; text-decoration-color: initial;"><li class="md-list-item" cid="n4" mdtype="list_item" style="box-sizing: border-box; margin: 0px; position: relative; display: list-item;"><p cid="n5" mdtype="paragraph" class="md-end-block md-p" style="box-sizing: border-box; line-height: inherit; orphans: 4; margin: 0px 0px 0.5rem; white-space: pre-wrap; position: relative;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">Arduino 1.8.13</span></p></li><li class="md-list-item" cid="n6" mdtype="list_item" style="box-sizing: border-box; margin: 0px; position: relative; display: list-item;"><p cid="n7" mdtype="paragraph" class="md-end-block md-p" style="box-sizing: border-box; line-height: inherit; orphans: 4; margin: 0px 0px 0.5rem; white-space: pre-wrap; position: relative;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">megaTinyCore 2.1.5 via Boards Manager</span></p></li><li class="md-list-item" cid="n8" mdtype="list_item" style="box-sizing: border-box; margin: 0px; position: relative; display: list-item;"><p cid="n9" mdtype="paragraph" class="md-end-block md-p" style="box-sizing: border-box; line-height: inherit; orphans: 4; margin: 0px 0px 0.5rem; white-space: pre-wrap; position: relative;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">Chip Settings: Clock 20MHz, default timer, Support serial event: No,  BOD 1.8V, BOD Mode: Disabled, EEPROM Retained, SUT 8ms</span></p></li><li class="md-list-item" cid="n10" mdtype="list_item" style="box-sizing: border-box; margin: 0px; position: relative; display: list-item;"><p cid="n11" mdtype="paragraph" class="md-end-block md-p" style="box-sizing: border-box; line-height: inherit; orphans: 4; margin: 0px 0px 0.5rem; white-space: pre-wrap; position: relative;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">ATtiny412F on SOIC-8 to DIP-8 adapter</span></p></li><li class="md-list-item" cid="n12" mdtype="list_item" style="box-sizing: border-box; margin: 0px; position: relative; display: list-item;"><p cid="n13" mdtype="paragraph" class="md-end-block md-p" style="box-sizing: border-box; line-height: inherit; orphans: 4; margin: 0px 0px 0.5rem; white-space: pre-wrap; position: relative;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">Programmer UPDI connected directly to target UPDI pin (PA0)</span></p></li></ul><figure  [... cut ...] _NO_TARGET_POWER</code></span><span md-inline="plain" class="md-plain" style="box-sizing: border-box;"> </span><span md-inline="strong" class="md-pair-s " style="box-sizing: border-box;"><strong style="box-sizing: border-box;"><span md-inline="plain" class="md-plain" style="box-sizing: border-box;">Workaround:</span></strong></span><span md-inline="plain" class="md-plain" style="box-sizing: border-box;"> Use 5V power pin or switch jumper to HV mode (may require 2 programming attempts)</span></p>

dduehren, 2021-06-07:
Thanks for the reply.  Here is my setup.

1.	The HV programmer is the one that you sell, though I built it myself.  I have tried it in both the HV and PHV modes without success.   I used it the way it’s pictured with power, gnd and the hvupdi signals going to the target.  No other power source was used.  The 1604 has a bypass cap.  There are other circuits on the target board.  There are no additional series resistors on PA0 (updi).
2.	When it didn’t do anything, I was concerned that I hadn’t built it properly.  I have a USB scope from BitScope but it doesn’t show accurate voltage levels, or I haven’t figured it out. So while I saw a pulse, I couldn’t tell if it were 12 Volts.
3.	That prompted me to build a hybrid with your design and the DIY HV UPDI programmer.  Instead of  the LCt1262CN8, I used a 12V external supply, and I used a different opto-isolator/switch because I had these things around.  I tested the opto circuit statically to set the series resistor driven by D12 so that 35ma was driven to a resistive load.  I connected directly to the UPDI line at the header.  Looking at the circuit for your design, I figured it wouldn’t hurt.  In any case, when I got that circuit working, I tried again with the same result as in #1 above.
4.	As for your other points.  I am using a DC powered USB hub for the USB to the Nano.  I have tried both HV and PHV modes. I am sure the software is working because I see the blue flash and I have used the D12 signal to trigger my scope.   The 12V signal does fall quickly when the pulse ends.  I know there’s internal resistance to your programmer and the board has other 5V loads.   I have tried both embedded devices and stand-alone devices and have gotten similar results. 

 

5.	I have also successfully programmed the ATtiny 1604 in the embedded setting using jtagupdi without HV.  I bricked the device because I had the wrong part selected 1614, not 1604.  Seems like the programmer should do better than that.
6.	What does the bootloader have to do with this?  I shouldn’t need the bootloader for an embedded part.  I prototyped the code on an uno, and once working, just retarget the pins and download.
7.	Is being bricked by setting the wrong target something that can be fixed with HV programming, or does that do some other kind of damage?

 

*	David

 

*	Arduino 1.8.13

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

Success 1: Success or Fail toggles, may require 2 programming attempts

Fail: ...bad response to leave progmode command: RSP_NO_TARGET_POWER Workaround: Use 5V power pin or switch jumper to HV mode (may require 2 programming attempts)

Dlloydev, 2021-06-09:
Thanks for your detailed reply.

Overall, your preparation and connections are described OK, however I'm concerned with the UPDI circuit you're using. The UPDI signal at the target is quite sensitive to timing, delays, voltage levels and the programmer's ability to quickly switch off the signal to high impedance mode.

- The HV voltage level should be within **11.5-12.5V**
- If using an opto-isolator, it needs to be fast acting and the opto emitter's output should be directly connected to target UPDI. Use **D12** on the Nano instead of the signal at the header. The header signal has components in between intended for the internal charge pump which will interfere with the required opto led signal. 

[image]

If you're working a lot of IC's, it would be good to have access to a Power Debugger. I know this is very costly, but it works incredibly well and if it can't unbrick an IC, there's no recovery to be had. Note that with this debugger, the signals are quite strong and  buffered. The HV pulse is ultra fast and sharp (far beyond their published recommendation of 100µs min width as its only 30µs) and it is  directly multiplexed onto the target UPDI signal.

I've tested this fast multiplexed approach on my UPDI-Key design and it achieves similar results. Unfortunately, I don't have the appropriate equipment and time to produce some of these, but its fully open source, so maybe someone will give it a try.

dduehren, 2021-06-10:
Thanks, I will try with D12 instead of D21.  The opto I have is a FOD814 which I think has comparable performance to the TLP185.

My biggest problem is accurately measuring the amplitude of the pulse with my Bitscope device.  

Who makes a power debugger?

David

David Duehren

Dlloydev, 2021-06-10:
If you're using an external 12V power supply (and if its regulated) the voltage should be OK. However, unregulated 12V DC adapter might output 16V at no load which is far too high. I would check the supply voltage with a multi-meter. The voltage loss through the opto-transistor would be quite small, probably around 0.2V drop when switched on.
[
Power Debugger](https://www.microchip.com/developmenttools/ProductDetails/atpowerdebugger)

dduehren, 2021-06-10:
I’ve already measured it at 12v.  The only reason it could be lower, as I observed, is if the current requirement is more than 35ma, because the resistor controlling the current into the emitting diode sets that.  

David Duehren

dduehren, 2021-06-17:
I have not worked on the HV programmer since last email. I’m wondering about the behavior of jtag2updi.

I had successfully programmed a ATtiny 1604 and it’s been running just fine over the last week.
Tonight I tried to download an update to the code that was running.
But jtag2updi / aversive retuned the infamous RSP_ILLEGAL_MCU_STATE message.
The part isn’t really bricked because it is still running the old code. 

Does it just get confused, and there’s nothing really wrong with the part(s)?

And does the serial adapter version behave differently?  This is an extraordinarily frustrating experience with these parts.  Maybe I should buy a real programmer and forget this DIY stuff which is so problematic.  Any model you’d recommend?

Sent from my iPhone

David Duehren

dduehren, 2021-06-17:
Hi,

 

This is the setup that I have been using.  I have been using D12, albeit with a smaller resistor value that I found experimentally to get me to 35ma with the opto I’m using.

 

I’m interfacing to the HVUPDI signal on your HV programmer and assuming that the normal UPDI signal is coming through.  I haven’t verified that, but I don’t think my soldering is that bad.

 

*	David

 

 <https://user-images.githubusercontent.com/63488701/121363831-47b51000-c905-11eb-8350-5f463c58a538.png>

dduehren, 2021-06-17:
Hi,

 

Please disregard previous email about illegal mcu state.  The good part was
able to be reprogrammed.  Somehow I managed to disconnect VCC.

 

It's getting late.

 

Sorry to bother you.

 

David

Dlloydev, 2021-06-17:
No problem, glad you got it working.

closed 2021-06-17 by Dlloydev